tacman-fp7/handover #3: hand removal from point cloud with probabilist model of hand color

giuliavezzani, 2016-10-18:


giuliavezzani, 2016-11-03:
Even if hand removal from point cloud is executed by simply discarding "grayish"  points (points that satisfy R=G=B values), the filtered point cloud are very good:

Point cloud filtered with **hand filter**:
[image]

Point cloud filtered with **gray filter** and **spatial filter**:

[image]

giuliavezzani, 2016-11-03:
**Volume filter**, i.e. the filter who selects only points inside a ciylinder inside the three finger positions, must be tested more in detail.

pattacini, 2016-11-03:
Quite impressive results 😃

giuliavezzani, 2016-11-07:
**Volume filter to be re-designed**.

I tested some particular cases such as:

[image]

where the fingers are very close among each others.
In this case, the best computed circle has centere in the origin of the hand reference frame, instead of being a small circle bounded by the finger positions.
However, if the small circle had been computed, too many points would have been discarded.
It's better to understand how to **change volume filter or to adapt it** to the objects under consideration, if volume filter is really necessary.

giuliavezzani, 2016-11-08:
Added **ellips filter**: it's similar **volume filter** but instead of reconstructing a circle, we reconstruct a ellipsoid passing by the thumb, index and palm positions.
In this way we have more degrees of freedom: two axis lengths and  two different offsets.

giuliavezzani, 2016-11-10:
A new code structure has been designed and has being tested in these days.  An outline of the code structure is given by:

<p align="center">
[image]
<p>

Two modalities are available:
* **online**: the point cloud and the kinematic information are collected on the robot (or on the simulator) and the filtered points are sent to the localization module
* **offline**: the points are read from files and the filtered points are saved into a new file.

All points and results can be:
* **saved** in files
* **sent** through `thrift` service

Various option are available. Parameters and functionalities can be set and asked through `rpc port.` See `doxygen` documentation for more information [**to do**].

giuliavezzani, 2016-11-11:
**New thrift services** added to the new code structure. Code is now ready for a **intensive test** on the robot, in order to find (possibly) **final parameters** for filters.

closed 2016-11-11 by giuliavezzani