tacman-fp7/handover #6: Address the problem of difference of sensitivity between the fingertips

massimoregoli, 2016-10-24:
The objective is to find a proper scale factor for each fingertip that will be applied to the related tactile response in order to deal with different tactile sensitivity.

massimoregoli, 2016-10-31:
Since results in #4 weren't satisfactory, the problem has been addressed in two more ways. For the experiments the Domino yellow box and the Pringles confection were used (see wiki).
* Empirical adjustment: while the object was grasped, the sensitivity scale of the middle and index fingers were adjusted online to improve the grasp as mush as possible. This method turned out to be more effective than the one in #4.
* Real force learning: in this case the ATI force sensors were used to create a mapping between the raw tactile data and the real force using learning methods (all the surface of the fingertip was explored during the training). Then, while executing the grasp, the usual tactile respose was replaced by the estimate of the real force using such mapping. The performance was very poor during the transition phase, while after a few seconds a very effective grasp was achieved (the index and middle fingers were applying approximately the same force). The problem of the transition phase may be due to the fact that during the training process the precence of hysteresis affected the learning of low forces (more involved during the transition).

In conclusion, it may be a good idea to focus on improving the real force learning, which seems promising, and consider the empirical adjustment as second best.

massimoregoli, 2016-11-02:
It is worth trying to properly scale the tactile response of the index and middle fingers by analyzing the tactile response when the same pwm is applied to the motors that actuate the fingers.

pattacini, 2016-11-02:
Not only the same `PWM` waveform should be applied to fingers' motors but also the fingers need to contact the same environment to have meaningful feedback.

From the difference recorded between the two tactile responses we could figure out how to trim our code to account for different sensitivities.

massimoregoli, 2016-11-11:
For this task I put (alternatively) the index and the middle fingers of the black iCub's left hand in constant contact with a hard object:

[image]

For each finger, I applied for three times a step-up input voltage to the motor actuating the proximal joint (the distal joint position was set to zero) and I measured the tactile response after 6 seconds from the application of the voltage. The input voltage was set to 300 ticks, that is close to the actual working conditions of the index and middle fingers during the execution of the stable grasp.

Finger | Tactile response (trial 1) | Tactile response (trial 2) |Tactile response (trial 3) | Tactile response (average)
:---: | :---: | :---: | :---: | :---:
Index | 18.57| 20.31 | 19.92 | 19.6
Middle | 34.41| 36.02| 35.26 | 35.23

The ratio between the average tactile responses of the two fingers was used to properly scale the tactile data while running the stable grasp.

In order to quantitatively evaluate the stability performance (and compare the results with other sensitivity calibration methods) I used the following method:
* I considered the 6 objects in the picture:

[image]

* for each object, I carried out 3 stable grasps
* at the end of each grasp I introduced a perturbation, which consisted in shaking the hand by means of a sine wave signal sent as position reference to one of the wrist joints (amplitude 10 degrees, period 0.5 seconds, total duration 1.5 seconds)
* for each trial I gave a score of 1 if the object didn't move, a score of 0.5 if it moved without falling and a score of 0 if the object fell from the hand
* the final score is obtained summing up all the scores given at each trial (the maximum score is 18)

I used this procedure to compare two sensitivity calibration methods, the one just explained and the "manual" one, according to which the sensitivity is scaled online while the grasp is being performed. The related results are showed in the following table:

Method| No. of times the object didn't move | No. of times the object moved | No. of times the object fell | Score
:---: | :---: | :---: | :---: | :---:
Manual | 11 | 2| 5 | 12
Step-up analysis | 9 | 2 | 7 | 10

In conclusion, the step-up analysis doesn't seem to give better results than the simple manual method. Probably this is due to the fact that during the application of the step-up voltage the surface of contact is different from the one that is actually involved on average during the grasps (and sensitivity is strongly related to the specific surface on contact).

closed 2016-11-11 by massimoregoli

pattacini, 2016-11-11:
@massimoregoli could you provide more insights about the so-called "manual" online scaling, or a pointer where it is explained?

massimoregoli, 2016-11-11:
I referred to the method briefly explained in the second comment of this issue, I had called it "Empirical adjustment". Basically it is an iterative online method:

* I start setting the same sensitivity scale factors for the index and middle fingers
* I perturb the grasp by applying an increasing manual force to the object
* under the action of this force, at some point the more sensitive finger will lose contact
* at that point I slightly reduce its sensitivity scale, then I perturb the grasp again, and so on...
* I stop when the grasp stability doesn't improve any more over the iterations

pattacini, 2016-11-11:
Thanks for the procedure