tacman-fp7/handover #8: Train a GMM with the aim of rotating objects by changing the hand configuration

massimoregoli, 2016-11-02:
Train a Gaussian mixture model using stable grasps in which long objects are inclined. It will be used to change the hand configuration in order to rotate the object and facilitate the handover task

massimoregoli, 2016-11-10:
For this task I used the left hand of the black iCub.
The new Gaussian mixture model has been trained using the following 12 objects:

[image]

I visually detected 5 stable grasps for each object (60 in total).
Differently from the GMM trained previously a few variables have not taking into consideration, so the regression schema in this case is the following:

[image]

The difference of position between the index and the middle finger has not been considered as input because it changes considerably during the object rotation and could lead to instability. 
The distal joints may also cause instability during the transition phase and for now they have not been considered.
The trained model is composed of two Gaussian components:

[image]

The rotation of the object is caused by the actuation of the thumb abduction joint. The following plot shows how it is related to the hand aperture (the only input variable):

[image]
 
The following video shows the GMM regression in action. The objects are rotated starting from the usual stable pose:

[GMM Regression Video](https://massimo-regoli-iit.tinytake.com/sf/MTExMTIzNF80NDA2NTkx)

The objects were correctly rotated as expected, but the most slippery objects moved a little bit during the transition.

closed 2016-11-10 by massimoregoli

pattacini, 2016-11-11:
That's impressive! Very well done @massimoregoli