tacman-fp7/handover #10: extend MUPF to deal with vision and tactile measurements

giuliavezzani, 2016-11-03:


giuliavezzani, 2016-11-03:
Since we have a lot of measurements coming from vision (on the front of the object) and only 3 finger positions (2 on the back of the object), we should give more importance to tactile measurements.
**Idea**: extend MUPF to deal with two kind of measurements, with two different variances

giuliavezzani, 2016-11-03:
Here are some results:
* Correct localization:

[image]

* The point cloud without points on the top of the object allows several solutions:

[image]

* Tactile measurments on the back of the object help discarding outliers in pose estimation:

[image]

giuliavezzani, 2016-11-03:
**Idea**: problems in getting the top of object point cloud could be caused by **hand filter**. To be tested.

giuliavezzani, 2016-11-07:
After some tests, it turns out that if the point cloud is down-sampled enough and, for instance, up to 80 points are exploited in the MUPF, a difference in the variances of vision and tactile measurements is not necessary.

giuliavezzani, 2016-11-07:
Tactile recognition: we can run 4/5 implementation of the MUPF with different object models, so that to perform at the same time localization and recognition. 
It would be very impressive in a demo!
Reliability to be tested

giuliavezzani, 2016-11-07:
**MUPF tests summary**

MUPF has been tested with some collected point clouds, tactile data and simplified models, showing good performance.

I also noticed that:

* **Two difference variances** for vision and touch measurements are not required if the ratio between the two kind of measurements is small (`num_vision_points/num_touch_points=30`)
* **Point cloud** plays a fundamental role in reliability and accuracy of localization **focus on point cloud quality**
* Instead of setting a `downsampling` variable, we impose a **`maximum number of measurements`**, since the number of points coming from vision might change a lot.
* We could run in parallel multiple implementations of MUPF with different object models (4/5) and perform at the same time localization and **recognition**.  With different objects could work well. [**to be tested**]

pattacini, 2016-11-08:
Ho did you test **hypothesis 1**?

Suppose we are dealing with `num_touch_points=3` and `num_vision_points=90` so that the condition of hypothesis 1 is verified.

Now if we basically run the algorithm twice, first with 93 points and then with 90 points (discarding `touch_points`), I suspect that there won't be any big difference, leading to the conclusion that tactile measurements do not count enough.

Can tactile measurements be decisive just for their locations only?

giuliavezzani, 2016-11-09:
Let's consider this scenario:

[image]

 
Before discussing **hypothesis 1**, I would like to highlight that if we uniformly down-sampled the point cloud so as to use 90 points, the density of those points is quite small, considering the number of faces and the surface dimension of the object. 

When I used a **smaller variance** for tactile measurements, the final solution is compatible with the tactile measurements, but not with all the vision measurements:

[image]

 I performed some tests and I saw that different variances are necessary and not  damaging only if the number of points coming from vision is much bigger. 

Probably, the same variance value for vision and tactile measurements is suitable for this kind of scenarios, since a good localization is achievable even only with vision measurements.

In support of **hypothesis 1**, I show this result:

[image]

where the **same variance** is used for tactile and vision measurements. 
It seems that tactile measurements together with the good points coming from vision are enough to ignore the noise points on the robot hand (on the left of the point cloud).

giuliavezzani, 2016-11-11:
MUPF has been extended to work with vision and touch, with quite good results.
More intesive tests are necessary, but they lie outside this issue.

closed 2016-11-11 by giuliavezzani