tacman-fp7/handover #15: grasping pose approach (idea n1)

giuliavezzani, 2016-11-11:
Start design grasping pose in case we are exploiting grasping pose idea n1

giuliavezzani, 2016-11-24:
Code structure:

- [ x ] **_apriori_ poses** are read from the model
- [ x ] the **estimated object pose** can be read from a text file or received from [**localizer module**](https://github.com/tacman-fp7/handover/tree/master/in-hand-localization/src/localizer)
- [ x ] the _apriori_poses are **rotated and translated** so that to be overlapped to the object model in the correct pose
- [ x ] the new poses are **shown** on the robot **cameras**
- [    ] the new poses have to be evaluated for the grasping of the second hand [**to do**]

giuliavezzani, 2016-11-28:
How to **select** a good pose among all the _apriori_ poses?

1. **manipulability index**
2. **distance** from joints **limits**
3. **first hand avoidance**. 
         - simple idea: assume to have symmetric _apriori_ poses (see [#14](https://github.com/tacman-fp7/handover/issues/14#issuecomment-262912820) and the following figure).  Consider the poses grouped in 4 couples: 2 on the top (3,4), 2 on the right (5,6), 2 on the bottom (7,8) and 2 on the left (1,2) of the object. Considering the object pose in the robot hand, we could **discard those couples that are closer to the robot hand**.

 For example, in the following picture
[image]
couples **(1,2)** and **(7,8)** would be **discarded** since they are closer to the robot hand.

giuliavezzani, 2016-11-30:
The  pipeline:
- [**point cloud acquisition and  filtering**](https://github.com/tacman-fp7/handover/tree/master/in-hand-localization/src/in-hand-segmentation)
- [**object localization**](https://github.com/tacman-fp7/handover/tree/master/in-hand-localization/src/localizer)
- [**_apriori_ poses visualization**](https://github.com/tacman-fp7/handover/tree/master/grasping-pose-selection)

has been tested successfully on the robot ([c875309](https://github.com/tacman-fp7/handover/commit/c875309be710b90b9cd087bf45f13c6810e19532)).

lornat75, 2016-11-30:
Consider that some iCub are not black. For example the Darmstadt robot is probably blue. Therefore on the Darmstadt robot non colored objects may work and we will have to avoid objects that are blue instead. Just to keep in mind, for the moment this is not the major contribution of the work and can be simplified.

giuliavezzani, 2016-11-30:
Aren't the palm and the fingers of the hand always the same? For example, the Blacky and the Purply have the same hand colors.

giuliavezzani, 2016-12-05:
Every **pose** is represented by an **index**.
The starting indices are 0.0. Then:

* the **distance** of each pose from the robot **hand** position is computed and each index is decreased proportionally to such a distance.
* then, a possible **q vector** is computed by using the inverse kinematic for reaching the desired values. **NB**: in case they are not reachable, a different pose is taken into account in the q vector. For this reason, the **error** between the **ideal desired pose** and the **pose obtained by the inverse kinematics** is computed. Then, the **manipulability** of q together with the error is used to update the pose indices.

An example of such an approach is shown in the following figure.
The first robot hand (right) is ideally located on the bottom right corner. The selected pose (number 4), together with the pose indices values (in red the lower, in green the higher ones) seems to be consistent with the proposed scenario.

**NB** The color of the pose 0 should be "more red". Some tests on a real scenario (on the robot possibly) will help to test this approach.

[image]

giuliavezzani, 2016-12-08:
The code has been tested on the robot [[e595a62](https://github.com/tacman-fp7/handover/commit/e595a62ce3a5a9013df49d5cdc583bae28f89b4a)].

**Object localization**

[image]

See discussion in the [previous comment](https://github.com/tacman-fp7/handover/issues/15#issuecomment-263835179). 

**Selected poses**
The manipulability index is periodically computed for each pose, after solving every time the inverse kinematics. In this way, two possible poses are selected:
* pose 2:

[image]

* pose 6:

[image]

**Pros** :smiley:
* the selected poses are **reachable** by the right hand
* the poses **colors** (and then the indices values) correctly **distinguish** between reachable and not-reachable poses. For every couple (0,1), (2,3), (4,5), (6,7),  one of them is "reddish" and the other "greenish", according to the reachability of the orientation.

**Cons** :persevere: 
* the results of the inverse kinematics are **not constant**. The selected pose changes very often
* sometimes, the **errors** between the desired pose and the pose obtained by inverting the kinematics are **large** and, therefore,  the selected pose is wrong

In conclusion, even if we are on the right track, manipulability computation needs to be investigated.

pattacini, 2016-12-08:
Well done @giuliavezzani ✨ 

Are you sure that **pose 2** is reachable in that configuration? I don't know how the first hand is in the space, but assuming it in the configuration I say in the past days, it seems to me that **pose 2** is barely reachable by hand 2. I might be mistaken though.

Results that are not consistent over time are somehow expected, given that we deal with a **redundant problem**. There are several ways to constrain the optimizer to refine this variability.

What I don't understand here is if the optimizer is prone to fail for desired poses that you proved beforehand are attainable.

giuliavezzani, 2016-12-08:
I solved the mistake :smiley:
I didn't actually enable the **torso joints** (error in my code) and the poses computed by solving the inverse kinematics (called in the iCub documentation `xd_hat`, and `od_hat`) were very **different** from the desired ones (`xd`, and `od`).

I tested the new code on the simulator, replicating the scenario of the previous [comment](https://github.com/tacman-fp7/handover/issues/15#issuecomment-265692371). 
Hereafter, you can see the selected pose and all the other poses colored according to their reachability:

[image]

With the correct computation of `xd_hat` and `od_hat`, the **_greenish_ poses have zero position errors** (`xd - xd_hat`) and **_low_ orientation errors** (`od-od_hat`). The **reddish poses** have instead the same position error but **higher orientation errors**.

In this scenario, the **only selected pose** is pose number 5, since the correct solution of the inverse kinematics provides a better index for pose 5 with respect to pose 0. 
In particular, I show on the camera both the pose given by `xd_hat` and `od_hat`  and the pose given by `xd` and `od`. They are exactly the same. 

Furthermore, I also added a **control by the user** on pose selection procedure. When a pose is selected, the poses indices are not computed any more. The inverse kinematics is solved again, and then the poses indices and the selected pose are updated, only if the user is not satisfied with this pose and asks for a new pose.

pattacini, 2016-12-08:
Great debugging!

pattacini, 2016-12-08:
Well, hold on, you cannot really use the torso, because by doing so the first hand will start moving as well...

To ease the handover we should move the first hand towards a more comfortable configuration.

pattacini, 2016-12-08:
The fact is that with only 7 DOF, the dexterous workspace is fairly limited, unfortunately.

This leads me to think that it's likely we'll need the two hands approach...

@massimoregoli

giuliavezzani, 2016-12-28:
In order to check if the **two hand approach** is strictly necessary, I performed some further tests.
In particular, I collected a new **data set** putting the _first hand_ as much as possible near to the  **workspace** of _second hand_. 

 In the following example, the _first hand_ is the left hand. The object is located at **y=0.0 - y=0.1**, so that to be in the workspace of the _second hand_, i.e. right hand in this example:

[image]

Hereafter, I report the acquired point cloud, the object models in the estimated poses and the pose selected with the **torso DOFs enabled** and with the **torso DOFs blocked**. 

### **test no.1**

* point cloud and estimated pose:

[image]

* pose selected with **torso DOFs on**:

[image]

* pose selected with **torso DOFs off** (the selected pose is the number 3, but the actual pose computed with the kinematics is shifted, since it is not reachable without enabling the torso)

[image]

### **test no. 2**

* point cloud and estimated pose:

[image]

* pose selected with **torso DOFs on
[image]

* pose selected with **torso DOFs off**:

[image]

### test no. 3

* point cloud and estimated pose:

[image]

* pose selected with **torso DOFs on**:

[image]

* pose selected with **torso DOFs off**:

[image]

Test no. 1, no. 2, no.3 show the pose selected in 3 different poses, respectively nearer and nearer to the workspace of the second hand (right in this case). However, a good solution is not found without enabled the torso DOFs.
Hence, the **two hands approach is striclty necessary**.

closed 2016-12-28 by giuliavezzani