tacman-fp7/handover #16: check the status of the iCub in Darmstadt and make it ready to be used

massimoregoli, 2016-12-04:
- [ ] check if joints are working properly
  - consider replacing the tendons of the thumb, index and middle fingers even if they are not broken
- [ ] check if tactile sensors are working properly
  - verify also that they don't suffer hysteresis too much
- [ ] check the specific calibration of the joints and correct possible offsets (especially for distal and abduction-adduction joints)
- [ ] tune the parameters used for the grasp approach phase
- [ ] for each finger test the low-level force controller by tracking square/sine wave reference signals
  - try first with the PID gains used on the black iCub
  - if not working, try using manual tuning
  - if not working, carry out the plant identification (or consider replacing the tendon)
- [ ] test the high-level PID controller by tracking square/sine wave reference signals
  - try first with the PID gains used on the black iCub
  - if not working, try using manual tuning 
- [ ] tune the sensitivity scale factors while grasping an object
- [ ] test the stable grasp based on Gaussian mixture model
  - introduce an external perturbation to evaluate stability
- [ ] test the object rotation task

giuliavezzani, 2016-12-04:
What about creating a new project called **missions** or something like that and put all the issues dealing with it there?

giuliavezzani, 2016-12-13:
@massimoregoli, if I have time, I can try to perform some of these checks. I'll let you know after my first day in Darmstadt (15th) and, in case, I'll ask you for more details on the tests to be performed.

massimoregoli, 2016-12-14:
Thanks, anyway I hope to finish all of them by tomorrow, even if we will have to share the robot (unfortunately there was a lot of work for the technicians and the robot became available only a couple of hours ago)

massimoregoli, 2016-12-19:
All the tests of the list have been accomplished, the status of the iCub in Darmstadt is now the following (the description refers to both hands):
* All the joints and the fingertips work properly.
* Differences in the joints calibration have been taken into account in the code in order to keep the same grasp configuration of the black iCub.
* The grasp approach is now effective (parameters have been properly tuned).
* The low-level PID controllers' gains have been tuned in a simple way: the proportional and the integral gains were scaled by the same factor (so keeping the proportions). Such factor was chosen for each finger by looking at the controller performance while tracking a square wave reference signal. Despite its simplicity, this method turned out to be sufficient to reach a good level of force control.
* There was no need to tune the high level PID controller's gains. Indeed, the controller had good performances while tracking a sine wave reference signal and using the same gains of the black iCub.
* The different sensitivity of the fingertips is now taken into account in the code, thanks to that the stable grasp using the Gaussian mixture model resulted to be very stable while grasping a tennis ball and a box (even more than in the black iCub)
* The object inclination was correctly performed while grasping boxes of different thickness

closed 2016-12-19 by massimoregoli