tacman-fp7/handover #17: check iCub's vision system and test in-hand-localization in Darmstadt

giuliavezzani, 2016-12-04:


giuliavezzani, 2016-12-04:
- [ ] if not done, install:      
         - [stereo vision](https://github.com/robotology/stereo-vision )
         - [segmentation](https://github.com/robotology/segmentation)
         - [CGAL](http://www.cgal.org/download/linux.html)
- [ ] test:
         - **iKinGaze**
         - **yarpmotorgui**
- [ ] calibrate **SFM** 
- [ ] parameter tuning of **dispBlobber**, starting from parameters found in [#12](https://github.com/tacman-fp7/handover/issues/12#issuecomment-262990625)
- [ ] test quality of point cloud acquisition and filtering, with parameter tuning of **`in-hand-localization` filters** (see [#12](https://github.com/tacman-fp7/handover/issues/12#issuecomment-262990625))
- [ ] test localization results
- [ ] test calibration between cameras and kinematics. If necessary, install [depth2kin](https://github.com/robotology/icub-main/tree/master/src/tools) and execute it

giuliavezzani, 2016-12-06:
@massimoregoli and @nawidj, I'd ask you to follow these steps to performe **fine calibration** of the `iCubDarmastadt01` head, since it may be required some mechanical tunings by julien.

Hereafter, you find **step-by-step instructions** (that julien already knows).

1. check in the `yarprobotinterface.ini` which is the used `.xml` (e.g. icub_all.xml);
2.  in that `.xml`, find the file concerning the head (e.g. head-cfw2_can0-mc.xml);
If the `iCubDarmstadt01` follows the standard of the robotology folder [**robots-configuration**](https://github.com/robotology/robots-configuration), the file of interest should be: 
[**hardware/mechanicals/head-cfw2_can0-mec.xml**](https://github.com/robotology/robots-configuration/blob/master/iCubDarmstadt01/hardware/mechanicals/head-cfw2_can0-mec.xml)
3. **NB**: you should not modify the standard file of their robot, but the **installed configuration file** in the **.local** folder. Thus, if they haven't install it, do:
       `yarp-config robot --import iCubDarmastad01`
and then, work in the **.local** repo
4. follow [these instructions](http://wiki.icub.org/wiki/HeadFineCalibration) to perform calibration.

If you need help for one of these step, please ask! :smiley: 
Likely, you will need julien's help for mechanical `eyes_tilt` calibration [head joint 3].

In the end, I'd ask you a picture of the iCub gaze :smiley:
@pattacini: is it ok this description?

Thanks guys!

pattacini, 2016-12-06:
👍 @giuliavezzani 

cc @julijenv

nawidj, 2016-12-08:
@giuliavezzani  we will perform the calibration when @julienv gets here. It looks like Alex has started installing the software you requested. We will be having a detailed look at the robot tomorrow.

giuliavezzani, 2016-12-08:
thanks @nawidj !! :smiley:

julienV, 2016-12-08:
not sure why i am getting referenced here, must be a typo ;)

pattacini, 2016-12-08:
Ya, definitely, @julienV 😉 
@giuliavezzani wanted to mention @julijenv instead.

paraschos, 2016-12-09:
@giuliavezzani 

We tested: Moving the right arm and tactile feedback from the right hand
The left arm has a broken tendon and therefore we removed it from the conf.

The eyes return images from the cameras

MotorGui, iKinGazeCtrl runs

Stereo vision, segmentation, CGAL installed 

depth2kin runs 

One note: the camera calibrator reads a config file from icub-main/app/cameraCalibration/conf/icubEyes.ini. Shouldn't instead read from robots-configurations/iCubDarmstadt01/icubEyes.ini?

paraschos, 2016-12-09:
@nawidj configs and code can be found in /home/icub/software_shared

paraschos, 2016-12-09:
One more note: please do not add files in .config or .local. Instead use the git repositories on branch tacman.

giuliavezzani, 2016-12-09:
@paraschos thanks for your work!

Will the left arm be ok for next thursday?

Is there a file robots-configurations/iCubDarmstadt01/icubEyes.ini?
If yes, there should be some problems in the .ini files, because I think it should read from robots-configurations.

nawidj, 2016-12-09:
@giuliavezzani don't worry the arm will be fixed.

giuliavezzani, 2016-12-09:
Great! Thanks all! :smile:

paraschos, 2016-12-09:
@giuliavezzani Sure there is an iCubEyes.ini in the robots-conf repo. It would be great if you can compare the values and merge to two files. The reason that icub-main is being read is that we manually specify to read that one, no problem with the configuration files :)

giuliavezzani, 2016-12-13:
@julijenv, did you check the `tilt` of the eyes? 
I'll be ready for working in Darmastadt on the 15th, in the morning. If you'll be still there, we can have a look together. 
Otherwise, if you notice something wrong, could you fix it ? That's the only DOF I can't calibrate by software :smiley:

giuliavezzani, 2016-12-15:
- [ ]  `meshlab` install on both the computers for point cloud visualization.

massimoregoli, 2016-12-15:
@paraschos and @nawid, I had the following problems:

* I tried to launch `iKinGazeCtrl` and `iKinCartesianSolver` from **yarpmanager**, but I didn't manage to do it
* on the other hand, I managed to launch them from terminal
* If I launch them from terminal, other modules that require them are not able to find `iKinGazeCtrl` and `iKinCartesianSolver`, even if they are running.

Any hints or suggestions?

giuliavezzani, 2016-12-15:
Summary after *day #1*:

- **iCub eyes** have been calibrated:

[image]

- handover code has been installed
- handover code is running
- `SFM` has been tested and calibrated [more or less]
- `dispBlobber` parameters have been updated with the good ones

**NB**: performance are very poor so far, need to be improved tomorrow.

giuliavezzani, 2016-12-16:
Basic functionalities of the vision system have been restored.

- Cameras outputs:

[image]

- `SFM` and `dispblobber` outputs:

[image]

- blob selected by `in-hand-segmentation` module:

[image]

pattacini, 2016-12-16:
Great achievement 😄

giuliavezzani, 2016-12-16:
Fine arm calibration executed.

This is the robot in the [**desired configuration**](http://wiki.icub.org/wiki/ArmFineCalibration):

[image]
[image]
[image]

giuliavezzani, 2016-12-16:
## Summary of first tests on **`iCubDarmstadt01`**

###  Overview
[Here](https://github.com/tacman-fp7/handover/issues/17#issue-193362435) you can see the checklist of what I managed to do.
Just a few comments about the achieved goals:

- **arms and eyes calibration** has been performed, but they might be needed to be redone, considering the low performance and obtained results.

- the **installed code** (handover code and  `dispBlobber`) is correctly running

- it's quite difficult to correctly calibrate **`SFM`**, even if, finally, I managed to do it. For more details about this issue, see [#22](https://github.com/tacman-fp7/handover/issues/22)

### Results

Even if **`SFM`** was very unstable, I managed to get some point clouds.
- In particular, I could test **coarse filter** and **hand filter**. In the following figure, all the points are the output of the coarse filter, wherease the blu points are the output of the hand filter.
[image]

- unfortunately I didn't get good results for the **ellipse filter**, since it strongly relies on the fingers positions, that were, however, totally inaccurate. In this figure, the output of the cameras consists of the colorful points, whereas the 3 white points are the fingertips positions. They are not consistent...
[image]

Given the huge offset between kinematics and camera (even after the fine calibration of the arms), I tried to perform `**depth2kin**`. Sincerely, I don't know if depth2kin could be enough, or if the robot actually need a mechanical calibration..
However, I didn't manage to perform depth2kin due to the following reasons:

- **`SFM`** instability
- **torso oscillations** (see [#21](https://github.com/tacman-fp7/handover/issues/21))

All the points were discarded by the robot, the hand segmentation was unreliable and the robot swang too much, so that I decided to stop with the experiments.

giuliavezzani, 2016-12-30:
### Tests will be performed when the `iCubDarmstadt01` will be @IIT

closed 2017-01-26 by giuliavezzani