tacman-fp7/handover #22: [Reminder] problems in running cameras

giuliavezzani, 2016-12-16:
Sometimes, when I try to launch cameras, one of them is not working (e.g. left).
If I play with the `yarpmotorgui` changing the value of `eye version` of the left, I got some strange images on the `left` camera.
The same happened also to the right.

But, sometimes they can be launched without any problems.

pattacini, 2016-12-16:
@giuliavezzani it's likely then there's a problem with the **firewire cable** connecting the camera to PC104. Due to wear and tear and the fast movements of the bulb it tends to become loose.

giuliavezzani, 2016-12-16:
thanks @pattacini. I'll leave here this issue just as a reminder for the next time, when technicians will be here again :smile:

lornat75, 2016-12-16:
I think the problem is in the camera flat cables, we can bring replacements next time, they are fairly easy to mount.

Sent from mobile device

-------- Original Message --------
Subject: [tacman-fp7/handover] [Reminder] problems in running cameras (#22)
From: Giulia Vezzani
To: tacman-fp7/handover
CC: Subscribed

Sometimes, when I try to launch cameras, one of them is not working (e.g. left).
If I play with the yarpmotorgui changing the value of eye version of the left, I got some strange images on the left camera.
The same happened also to the right.

But, sometimes they can be launched without any problems.

giuliavezzani, 2016-12-16:
Though this morning I managed to calibrate `SFM` and got quite good disparity map (see [#17](https://github.com/tacman-fp7/handover/issues/17#issuecomment-267567631)), now I can't calibrate `SFM` any more.
 I tried several scenarios: a lot of different objects on a table, a 3d scenario in the room, different SFM parameters, different cameras tuning etc.., but calibration never converges. In addition, when I **rarely** managed to calibrate `SFM`, the resulting point clouds are actually unrealistic and wrong.

lornat75, 2016-12-16:
Are there enough features in the images?

giuliavezzani, 2016-12-16:
Yes, even when I have a lot of features. 

However, the behaviour is **not constant**. Now, I manage to calibrate SFM, but the obtained point cloud are completely crazy. 

Furthermore, I've also noticed:
-  that the cameras need very **different values** in the `grabber remote GUI` in order to have similar images. 
-  a difference between the **iCub eyes**: the cameras are not perfectly aligned in the eyeballs. In the following picture, I'm trying to show this difference:

[image]

 I don't know if all these elements can have any influence on SFM, mostly since yesterday, in the same scenario, I got better results. 
I need to discover the source of such a lack of robustness..

giuliavezzani, 2016-12-16:
## Summary of all problems experience with cameras

1. some problems in **cables connection**. Sometimes the cameras need to be moved a little bit to actually be connected and provide images

2. the required **cameras parameters** for having similar image are **much different** between left and right 

3. problems in **`SFM`** **calibration and reliability** (even if, after all, I think I managed to find an acceptable calibration - at least for point cloud acquisition)

closed 2017-01-26 by giuliavezzani