tacman-fp7/handover #23: Allow the execution of the stable grasp on both hands at the same time

massimoregoli, 2016-12-30:
At the moment the stable grasp can be run only on one hand at a time to avoid conflicts between ports. Change the code in order to allow its execution on both hands at the same time, as needed for the handover task.

massimoregoli, 2017-01-05:
Now two instances of the stable grasp module (one for the left hand and one for the right hand) can be run at the same time. See [here](https://github.com/tacman-fp7/handover/wiki/Stable-grasp-module-interface) for more details.

closed 2017-01-05 by massimoregoli

pattacini, 2017-01-05:
Well done @massimoregoli 

Also, I've seen you provided the documentation. Normally, this kind of info should accompany the code itself, as done for example in [**how-to-document-modules**](https://github.com/robotology/how-to-document-modules) 😉

pattacini, 2017-01-05:
Ah ok, it's a partial documentation for the specific task.

massimoregoli, 2017-01-07:
Yes exactly it is specific :blush: