tacman-fp7/handover #24: Investigate the possibility of increasing the thumb strength

massimoregoli, 2016-12-30:
At the moment the poor strength of the thumb represents the bottleneck of the whole grip strength. Investigate how it can be increased, if possible.

massimoregoli, 2017-01-07:
Good news, it's definitely possible to increase the grip strength! Basically, the actual pwm that is sent to the motor that actuates the thumb proximal joint is always quite low (something like 200-300) even if the maximum pwm (1333) is requested from my module. This is because at the firmware level we find the following line:
* `actualPWM[thumbProximal] = requestedPWM[thumbProximal] - actualPWM[thumbAbduction]`

So the requested pwm at the thumb proximal joint is corrected according to the actual pwm at the thumb abduction joint, since such joints are coupled. To exploit all the power of the thumb one solution (we can talk about other solutions when we meet) could be to broaden the pwm range of `requestedPWM[thumbProximal]` from [-1333 1333] to [-2666 2666]. I already tried to do that and the grip strength was actually increased.
Before performing more tests there is still a little problem to solve firmware-side: I noticed that the value of `actualPWM[thumbAbduction]` is truncated into the range [-1333 1333] only AFTER that the above formula is applied, while it should be done before, otherwise when saturation occurs at the abduction joint, the value of `actualPWM[thumbAbduction]` won't be the "actual" one, and the formula won't give the correct value of  `actualPWM[thumbProximal]`. This is important in our problem because when I broadened the pwm range of `requestedPWM[thumbProximal]` to [-2666 2666] and the thumb started pushing more, I saw that saturation actually occurs at the thumb abduction joint, leading the value of `actualPWM[thumbProximal]` to be lower than it should (so the grip strength was increased but not as much as wanted).
Unfortunately when I noticed this problem there wasn't anyone who could compile the firmware code, it will be done on Monday, then I will perform more tests in order to:
* check if the PID performance related to the thumb proximal joint is still good after the change in the pwm limits,
* find a new value of grip strength that will allow to have a stable grasp for most objects without breaking the tendon (I guess that the tendons weakness will be our new concern),
* evaluate the related grasp stability using the method explained [here](https://github.com/tacman-fp7/handover/wiki/Grasp-stability-evaluation).

pattacini, 2017-01-08:
Good catching.

massimoregoli, 2017-01-10:
The firmware has been changed, I tested it and the mentioned problem has been solved.
About the stability performance, we have to talk separately about the following phases:
* **transition phase**: since the fingers get in contact with the object, it takes a while before the grasp stabilizes to the desired grip strength (simply because the pwm sent to the motor actuating the thumb proximal joint is now in a wider range and a higher grip strength can be achieved) . Until that moment the grasp may be quite unstable. I guess that this problem can be faced by playing with the P.I.D. gains, or more simply, scaling the pwm by a constant factor.
Another problem is that now, given the higher strength, it's more likely that the thumb slips along the object. Such slippage could be stopped by fixing the position of the thumb abduction joint. The problem is that now saturation occurs at that joint, so basically the motor is not able to counteract the slippage movement, which leads to instability. In my opinion the only way to solve this problem is to choose not slippery objects (or to make them so by changing their surface).
* **grasp stabilized**: once the desired grip strength is reached, the grasp becomes actually very stable. I run the [stability evaluation method](https://github.com/tacman-fp7/handover/wiki/Grasp-stability-evaluation) and I achieved the maximum score (18/18), so once the grip strength was reached, none of the objects of the set fell from the hand when a perturbation was introduced. Notice that as I did in the past when evaluating grasp stability, I always wait that the grip strength is reached, so I basically ignore the transition phase.

closed 2017-01-10 by massimoregoli