tacman-fp7/handover #25: Tests on the robot

giuliavezzani, 2017-01-11:
This is to collect tests performed on the robot

giuliavezzani, 2017-01-11:
### Online Tests on the Blacky: ToDo List

- [ ] install new code and suitably update the config files (robot name, online mode etc..)

- [ ] **fine** arms and head **calibration**

- [ ] **depth2kin** calibration

- [ ] test @massimoregoli's code in order to select at least two objects (sugar box for sure)

- [ ] acquire and check **point clouds** of the two objects in **left hand** and **right hand**

- [ ] pose selection and closed chain **without moving the second hand** with the box

- [ ] pose selection and approach **moving the second hand** with the box

### Offline Tests: ToDo List

- [ ] analyze **point clouds** collected with the object in the **right hand**

- [ ] test pose selection and closed chain with the **left hand moving**

- [ ] annotate the **new** selected **object** model with 8 **poses**

- [ ] test **pose selection** and closed chain with the **new object** on the simulator (and if possible, on the robot)

giuliavezzani, 2017-01-13:
## Results 1.0
It's useful to summarize out pipeline weaknesses:

### External modules

- **SFM** lacking robustness

- **depth2kin** turns out to be necessary - too big errors between kinematics and vision

- **depth2kin** cannot be launched (see [icub-main #409](https://github.com/robotology/icub-main/issues/409))

### Internal modules

- we need a  very **stable** grasp of the first hand, i.e.:
       1. fingers moving as less as possible
        2. object surface that can be seen from the robot viewpoint should be the maximum one

giuliavezzani, 2017-01-16:
## Results 1.1

Here are the results and comments of the [tests](https://github.com/tacman-fp7/handover/issues/25#issuecomment-271921011) performed on the robot.

### Online tests on the robot:

- **SFM** is not robust but, even if sometimes the disparity map looks quite awful, the point cloud are not so bad 

[image]
- **depth2kin** can be now compiled (but I have not managed to test it yes)
- @massimoregoli's code is now pretty good :smile: The **grasp is stable** and I managed to get a quite extended point cloud:

[image]
 - **pose selection** with closed-chain. In the following picture, the **selected pose** is no. **6**. The pose on the right is the final pose to be reached by the right hand. At the same time, the left hand should move so that the highlighted pose no. 6 attached to the object model totally overlap with the pose on the right.

[image]

- an example of (failed) handover test: [**video to be uploaded**]

#### Comments on online tests

- a robust **approach** design is fundamental (see the previous video)
- we can try moving the **two arms separately**
- code debugging to be done
- one good point :smile: : the grasp designed by @massimoregoli is very stable!

### Offline tests

- handover code works fine also with **right hand as first hand** on the simulator

giuliavezzani, 2017-01-23:
## Results 2.0

I tested the [**new approach**](https://github.com/tacman-fp7/handover/issues/27#issuecomment-273775015) strategy on the robot.
Unfortunately, I **didn't get point clouds good enough** to correctly localize the object. Thus, I couldn't properly test my code.
However, I'm quite confident that, **once a good localization will be provided**, **the proposed approach** **is likely to work.**

Hereafter I listed all the problems I faced and the results that make me quite optimistic about the future.

### Problems

- **SFM** performance are **random**.  This is the best point cloud I got with `vergence=5`, calibrated cameras and calibrated SFM:

[image]

 I also noticed that **SFM** performance seems to be better for object on the table than objects in the robot hands (and then closer to the eyes). The following figure show disparity map while the robot is looking at the object (on the left), while the robot is looking at the object in its hand very close to the table (in the middle) and while the robot is looking at the object very close to its eyes:

[image]

- I had some problems in make the robot correctly reach the waypoints: the cartesian solver couldn't reach the correct orientation. I solved the problem by using the [**minJerkVelCtrlForIdealPlant**](https://github.com/towardthesea/karmaWYSIWYD/blob/master/src/karmaMotor/main.cpp#L612)

## (Quite) Good Results

- **Moving the robot arms separately** seems to be enough in make the robot prevent hitting the object with the second hand. The general pipeline could be:
      - put the **first hand in the final position**
      - move the **second hand** toward the **final position**, passing by the **waypoints**

- the best pose and the waypoints are correctly selected 
- all the introduced parameters seem to be enough to design a good approaching strategy

Here are some examples of selected poses and computed waypoints:
-  `n_waypoint=1`, `theta=-20.0`, `offst_x_final=0.03`, `offst_z_final=0.03`, `offst_x_approach=0.08`, `offst_z_approach=0.08`

[image]

-  `n_waypoint=1`, `theta=-40.0`, `offst_x_final=0.03`, `offst_z_final=0.03`, `offst_x_approach=0.08`, `offst_z_approach=0.08`

[image]

Finally, you can find an example of a sort of performed handover at [this link]() (it actually worked by chance.

###  :exclamation:  Warning :exclamation: 

In my option,  the main point to be addressed is to **improve point cloud quality**.

pattacini, 2017-01-23:
Hi @giuliavezzani 

Regarding the use of `setTaskVelocity()`, there must be still something else.
This method, in fact, does also call in turn the solver, hence direct calls to solver should work as well.

giuliavezzani, 2017-01-23:
From what I understood, [the function  I used as example](https://github.com/towardthesea/karmaWYSIWYD/blob/master/src/karmaMotor/main.cpp#L612) accomplishes the following steps:
- a direction of movement is decided: `dir`
-  the error `e` between the current position and the desired one is computed, by using information on the positions coming from the  `getPose()`  function
- then, the function `minJerkVelCtrlForIdealPlant::computeCmd()`  provides the joints velocities necessary for moving the arm, once they are properly multiplied for the desired approaching direction
- then, those velocities are imposed with the `setTaskVelocities()`
These steps are executed until the norm of the error is lower a threshold.

Maybe I misunderstood how this code and the related functions work, but actually the arm behaves as I want: the desired positions are reached moving the arm along the desired direction and without changing the initial orientation.

pattacini, 2017-01-24:
`minJerkVelCtrlForIdealPlant::computeCmd()` provides velocities in the operational space, not in the joint space. It's a mere filter. Then these velocities are translated in the joint space through `setTaskVelocities()`, which in turn calls the solver underneath. That's why I was saying it must work also with simple calls to the solver.

pattacini, 2017-01-24:
Anyway, if it's working with `setTaskVelocities` is fine 😃 
Let's not loose time debugging other stuff.

giuliavezzani, 2017-01-24:
## Results 2.1

Some updates after new tests on the robot;

- **SFM** with **`sgbm`** option  is quite good (and then object localization is better than the previous tests). This is an example of the disparity map for the object:

[image]

- I managed to obtain some reasonable movements, even if handover was not perfect:

[image]

[Click here]() for the video of the best handover I obtained

## How to Improve

1. Intensively test **`localizer`**
2. **`depth2kin`** calibration
3. automatic check on the **blob**
4. improvements in **iCub movements**
5. tune approaching strategy **parameters**

closed 2017-02-09 by giuliavezzani