tacman-fp7/handover #33: Preparation for the final demo

giuliavezzani, 2017-02-13:
This is to organize the final rush for the demo.

giuliavezzani, 2017-02-13:
## What to Do

- [ ] perform **`depth2kin`** for the `right hand`
- [ ] perform **`depth2kin`** for the `left hand`
- [ ] check **point cloud quality** after `depth2kin`
- [ ] add control on **`context`** where necessary
- [ ] add **initial pose also for second arm**
- [ ] add **final movement** moving the **first hand** far away
- [ ] add gaze tracking ?
- [ ] automatize everything with **`lua`**
- [ ] find good parameters for **approach**
- [ ] check on number of measurements in `localizer`
- [ ] set **`trajectory time`** everywhere

giuliavezzani, 2017-02-14:
## One of the first  accomplished handover

[Here](https://www.youtube.com/watch?v=rGU-uDTCyGw) you can find a link to a video showing one of the first accomplished handovers.

So far:
- the `gaze interface` has been introduced to look at the hand poses
- several `rpc commands` are available to have nice poses at the beginning and the end and to play with the task

This is a good starting point

lornat75, 2017-02-14:
Well done!

giuliavezzani, 2017-02-14:
Here are other two (older) examples of handover:

- from the [right to the left hand](https://www.youtube.com/watch?v=2mDyMMvsGx8)
- from the [left to the right hand](https://www.youtube.com/watch?v=qbxKlZ2L0SE)

but without any gaze control.

giuliavezzani, 2017-02-18:
## Handover RtoL (Right to Left arm) Tests with  sugar box

I found a good initial position and the following parameters for performing **5/7 successful handovers**, using every time a different point cloud and estimated pose.
The method seems to be quite robust for the same initial pose, even if no compensation between kinematics and vision is performed.

The parameters are the following:

- `acqusition poses right`  joints :+1: 
```
-61.0 48.0 50.1 40.0  -31.0  17.0 18.0 
```
- `fixate point`:
```
y-= -0.18
z+= 0.15
```
- `offset_x_final`=0.03
- `offset_z_final`=0.0
- `offset_x_approach=0.08`
- `offset_z_approach=0.04`

lornat75, 2017-02-18:
👍

giuliavezzani, 2017-02-18:
Unfortunately, it's very difficult to have some generic and robust in very different situations, because of the strong non-linear offset between kinematics and vision ...

closed 2017-02-23 by giuliavezzani