tacman-fp7/handover #36: Handover with 4 objects

giuliavezzani, 2017-03-14:
This is to summarize the work for the paper

giuliavezzani, 2017-03-14:
Hereafter, I summarize the results obtained with **4 objects** for some main steps of the handover:

1. **point cloud** acquisition
2. object **localization**
3. **pose selection**
4. **kinematics-vision** offset compensation

## Objects taken into account:

1. Domino box
2. Jello box
3. Pringles tube
4. Mustard box

**Note** I focus on this bigger objects. If possible, I'll try to make the handover work also with the small red cup and the tomato soup.

## Annotated models
- 8 poses for object 1,2 and 4
- 24 poses for object 3 (due to the higher number of simetry axes)

## Point cloud & Estimated object poses 

[image] 
[image]
[image]
[image]

## Pose Selection
### Domino box, pose 1
[image]

### Domino box, pose 2
[image]

### Jello box 
[image]

### Pringles tube

[image]

### Mustard box, pose 1
[image]
### Mustard box, pose 2
[image]

## Kinematics-vision offset

Main points:

- Without any compensation, the offsets are huge
- I collected **2 poses** for the domino box (one per each of the usually selected poses)
- I used these 2 poses for correct the pose for the **domino** and the **jello**
- For the domino it worked **very well**
- For the jello it worked
- I need to test it with the mustard and the pringles, as soon as the grasp of these two objects will be stable (I'm waiting the help of @massimoregoli ).

Hereafter, the tests I performed with the domino and the jello boxes:

### Domino, pose lateral

- the pose with the lower z value is the corrected one
[image]

- handover successfully achieved
[image]

### Domino, pose on the top

- the pose with the lower z value is the corrected one
[image]
- handover successfully achieved
[image]

### Jello

[image]

# What to Do

- to test **2 poses** for kinematics-vision compensation on **mustard** and **pringles**. 
- improve **stableGrasp** for the 4 objects under consideration

giuliavezzani, 2017-03-15:
Hereafter I report some pictures of successful handovers with 4 objects (even if **stableGrasp** still need to be improved)

## Pose Selection

### Mustard
[image]

[image]

- The same happens for the **pringles tube** (I forgot to collect an image)

## Successful Handovers with the same correction Matrix
[image]
[image]
[image]
[image]
[image]

# Idea
I think that the successful use of a single correction matrix depends on the initial pose of the first hand during point cloud acquisition.
This idea will be tested in next tests

giuliavezzani, 2017-03-16:
# Tests on kinametics-vision compensation

If the initial pose is the same, **one compensation matrix** is enough and effective for all the objects.
Hereafter, you can see the corrected pose reached by the second hand for 5 objects:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

lornat75, 2017-03-16:
👍

giuliavezzani, 2017-04-04:
We managed to perform the handover with 5 objects :tada:

closed 2017-04-04 by giuliavezzani